lvgl/lv_drivers #319: Doesn't lv_drivers support lvgl9.2 yet?

lanistor, 2025-01-30:
`lv_disp_drv_t` used in lv_drivers, lvgl doesn't support anymore.

kisvegabor, 2025-02-06:
Correct, LVGL v9 has
- built-in drivers
- and a cleanar driver interface. 

See [here](https://docs.lvgl.io/master/details/main-components/display.html) and migration guide from v8 to v9 [here](https://docs.lvgl.io/9.0/CHANGELOG.html#migration-guide)

lanistor, 2025-02-06:
@kisvegabor Thanks, i had runned the simulator. But i got a problem:

When the simulator started, the screen is cluttered, after i drag the window' edge to change it's size once, it become normally.

### Cluttered window at start
[image]

### After draged
[image]

And here is my init code, is there any code wrong?
```cpp
#include <stdlib.h>
#include <unistd.h>

#include <SDL2/SDL.h>
#include "lvgl/lvgl.h"
#include "drivers/sdl/lv_sdl_window.h"
#include "drivers/sdl/lv_sdl_keyboard.h"
#include "drivers/sdl/lv_sdl_mouse.h"
#include "drivers/sdl/lv_sdl_mousewheel.h"

#include "./project.h"
#include "fl_utils.h"
#include "modules/setting/device/service.h"

#define MONITOR_HOR_RES     1024
#define MONITOR_VER_RES     758

static void hal_init(void);
static void hardware_init();

extern void flicker_demo_widgets();

static uint32_t sys_tick(void) {
  return (uint32_t) (esp_timer_get_time() / 1000ULL);
}

int main(int argc, char** argv) {
  (void)argc; /*Unused*/
  (void)argv; /*Unused*/

  /*Initialize LVGL*/
  lv_init();

  lv_tick_set_cb(sys_tick);

  /*Initialize the HAL (display, input devices, tick) for LVGL*/
  hal_init();

  hardware_init();

  flicker_demo_widgets();

  lv_refr_now(NULL);

  while (1) {
    lv_timer_handler();
    usleep(10 * 1000);
  }

  return 0;
}

/**
 * Initialize the Hardware Abstraction Layer (HAL) for the LVGL graphics library
 */
static void hal_init(void) {
  // monitor
  lv_display_t* disp_drv = lv_sdl_window_create(MONITOR_HOR_RES, MONITOR_VER_RES);
  lv_display_set_rotation(disp_drv, LV_DISPLAY_ROTATION_0);
  lv_display_set_color_format(disp_drv, LV_COLOR_FORMAT_RGB565);

  /* display buffer */
  int display_pixel_size = MONITOR_HOR_RES * 160;
  static lv_color_t* buf1 = (lv_color_t*)heap_caps_malloc(
    display_pixel_size * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
  static lv_color_t* buf2 = NULL;
  lv_display_set_buffers(disp_drv, buf1, buf2, display_pixel_size, LV_DISPLAY_RENDER_MODE_PARTIAL);

  // Input group
  lv_group_t* g = lv_group_create();
  lv_group_set_default(g);

  /* mouse */
  lv_indev_t* mouse_indev = lv_sdl_mouse_create();
  lv_indev_set_long_press_time(mouse_indev, 1000); // long press time

  /* keyboard */
  lv_indev_t* kb_indev = lv_sdl_keyboard_create();
  lv_indev_set_group(kb_indev, g);

  /* mousewheel */
  lv_indev_t* enc_indev = lv_sdl_mousewheel_create();
  lv_indev_set_group(enc_indev, g);
}

void hardware_init() {
  device_setting_service::read_epaper_vcom();

  device_setting_service::pre_init();
}
```

closed 2025-02-07 by lanistor